stephan192/hoermann_door #2: PCB source

riogrande75, 2021-02-20:
I've been interrested for a while already in hoermann-hcp protocol, but this seems to be the most closest and finished project to fit my needs.
Is there a source where I can get this pcb boards?

stephan192, 2021-02-21:
Currently there is no source.
But if more people are interested i can setup one.

stephan192, 2021-03-03:
I don't know a source where to get the PCBs and i will not setup one.

stephan192, 2021-03-17:
I added gerber files to the repository but i'm not 100% sure if they are working.
I ordered my empty PCBs also at JLCPCB (without assembly) back in 2019 and i think i used the added gerber files but i'm not 100% sure.

@jmonsted if you plan to order boards can please give feedback if the gerber files are fine or not?

jmonsted, 2021-03-18:
Thanks Stephan. I found out that EasyEDA would load .sch and .brd files and already ordered that way. JLCPCB claims it's already manufactured :)
Time to source the parts...

JPM87, 2021-04-25:
Hi Stephan, thanks for sharing. I am not familiar with microcontrollers so can we replace the PIC16F15324 with PIC16F1503?

thomasdupas, 2021-04-26:
Very nice Stephan, even though I found this project way too late (I have an UAP1, with a 8266 interfaced with it on dev board; but instead of creating a PCB to do it properly I might as well use this PCB/project and eliminate the middle man)

@jmonsted since we're ~1 month further, did you succeed to source the parts? I was going over the BOM trying to match components, but not certain about some (for example SRN6045TA-, doesn't contain a value, but looking at the PCB picture it might be a SRN6045TA-220M due to the 220 print?), so I'd like to cross-reference if you're willing to share.

stephan192, 2021-04-28:
You're right L1 is a SRN6045TA-220M.
See https://github.com/stephan192/hoermann_door/blob/main/board/RS485Interface_sch.pdf

thomasdupas, 2021-04-28:
thanks for confirming @stephan192 👍,  I didn't check the pdf.
Both the sch schematics view in kicad as the pcb -> bom list showed "SRN6045TA-" (just checked again by doing a grep on a fresh checkout, to be sure it wasn't a rendering issue). Guess something got cut off perhaps at some point; I'll double check the pdf vs sch to be sure if nothing else less visible got clipped

thomasdupas, 2021-05-03:
@stephan192 I started altering the circuit/PCB, since the used DC-DC converter doesn't seem to be available at my sources. Is there an interest for a PR with this + BOM, or should I keep it in a fork?
Reason I'm asking, I'll probably add some more inputs/outputs to the esp8266 to make the same PCB usable for another motor I'm interfacing with, if there is no need for PR I don't have to split the circuit/PCB updates in to 2 stages.

And for my understanding, why did you split the code between the PIC and the ESP? Was it to have the basic "UAP" discovery etc always available even when the ESP doesn't have running code / gets reprogrammed with updates?

stephan192, 2021-05-04:
@thomasdupas I think it's better when you keep your changes in your own fork. My board is working 24/7 for 1.5years now with the selected DC-DC, so it's seems to be a good choice. Additionally my intention was to create a very minimal board without any additional features like extra IOs.

The reason why is split the code between PIC and ESP is the mainly the LIN-Syncbreak. Each message has to start with a syncbreak and i didn't find an easy solution to emit syncbreak with the ESP. Due to the fact that i'm very familar with PICs i choose to use two controllers.
@raintonr solved it with some tricks on an raspberry which also cannot emit synbreaks.
https://github.com/raintonr/hormann-hcp/issues/1
https://github.com/raintonr/hormann-hcp/issues/9

jmonsted, 2021-05-05:
I managed to get almost all of the parts from LCSC, except the SRN6045TA, PIC and SS25 that are coming from Farnell.

I filled out the BOM in EasyEDA and shared the project here: https://oshwlab.com/jesper_1127/hormann-interface

Also, i ordered five of everything, so if anyone wants a kit, i can spare a few. I'm in Denmark. I think the kit should run about €20-25 with shipping. Poke me directly if you're interested.

NemoN, 2021-05-05:
I'm interessted. I have some spare PCB too.

Abildgren, 2021-05-05:
@jmonsted, I'm in Denmark too, and is interested in two kits. Your github page does not leave details on how to contact me directly though.

thomasdupas, 2021-05-06:
my components just arrived; and PCB's are supposed to arrive tomorrow. I'm also willing to help out other people if there is a demand (located in Belgium), I have 2 spare PCB's + components for them

NemoN, 2021-05-06:
@thomasdupas I would be interested in a component set (with or without PCB - I had some made myself) Would you like to contact me via email (thomas.oeding@gmail.com)?

i5Js, 2021-05-23:
@jmonsted I’m living in The Netherlands, do you still have one?

Thanks,

thomasdupas, 2021-05-23:
@i5Js I can only provide complete boards most likely since I deviated from the original design. I currently ran out of PIC,ADM485’s and inductors; but just ordered new ones which should arrive in a week, when I could supply you with some boards.
Currently only 1 remaining completed board, which is already promised to @NemoN .
Drop me a mail if you need more information, my mailadres should be on GitHub profile page (if not, it’s <first name>@<lastname>.be

jmonsted, 2021-05-23:
I made a mistake and got the PICs in the wrong package, so i'm waiting for those to arrive (on tuesday). I have three kits up for adoption :)

I'll add my email to my github profile.

closed 2021-07-26 by stephan192