stephan192/hoermann_door #3: Error "6." after connecting board to SupraMaticE2

riogrande75, 2021-05-13:
I recently built this project and I am pretty confident, that everything was soldered together and programmed correctly.
Right now, a UAP1 and a button is connected to my SupraMaticE2 - this works so far.
As soon as I replace the UAP1 with the board from this project, I get following output on my MQTT server:
```
[2021-05-11 16:39:46.695584] garage_links/cover/state stopped
[2021-05-11 16:39:46.695644] garage_links/sensor/garage/status offline
```
Unfortunately it does not respony to any of my issued commands:
```
[2021-05-11 16:41:12.825838] garage_links/sensor/garage/set CLOSE
[2021-05-11 16:41:18.133533] garage_links/sensor/garage/set TOGGLE
[2021-05-11 16:41:59.022011] garage_links/sensor/garage/set OPEN
[2021-05-11 16:42:01.973374] garage_links/sensor/garage/set OPEN

```
If I want to open close the door manually with the button, I just get fault "6." in the display. This points to a faul on the "closed circuit".
As soon as I put UAP1 back in place everything works fine again.
Connecting RPi with RS485 and running [https://github.com/raintonr/hormann-hcp](url) works too, so I guess the door and control unit are ok.

Any idea from the specialists on how I could debug here?

stephan192, 2021-05-15:
My guess:
A wrong, missing or twisted part is messing up the hoermann bus.

To narrow down the problem:
1. The manual for my Supramatic E3 says error 6 means "Systemfehler" (system error) which would make more sense.
2. When you have disconneted all slaves (UAP1, this board, raintor's interface) the drive shows error "7". Right?
3. `garage_links/cover/state stopped` is the default state if no communication with the drive could have been achived.
4. `garage_links/sensor/garage/status` offline is the default state if no BME is connected.

The next steps i would perform:
1. Check the board's supply voltages. Are 24V, 5V and 3.3V fine?
1. Check the RS485 lines with an oscilloscope. Can you see the master messages or is it just a flat line?
2. Check the board parts on the RS485 side (IC4, T1, T2, T3, T4 + surrounding parts). My guess is that one or more parts have wrong values or are mounted twisted.

closed 2021-07-26 by stephan192

riogrande75, 2021-12-09:
Ok, finally found time to work on this issue again. Factory resetting my SupraMatic E2 solved the error 6 problem easily.
BUT: I dunno why I was reading over it so many times: This project is for Supramatic**E3** and not my E2 - so it will not work anyway. :-(