stephan192/hoermann_door #4: Toggle lights

snoekieboe, 2021-05-21:
Hi Stephan, First of all thanks for all you nice work. I've managed to get my hand on a PCB made according to your design and it is working like charm.

I've noticed the presence of a toggle light option in the Pic16 code and was wondering if it is already possible to use this feature through MQTT... could not verify this to be true whilst looking through the esp8266 code.

Thanks in advance !

stephan192, 2021-05-22:
Hi snoekieboe,
as i analyzed the bus i found out that one bit toggles the light, so i implemented it in the pic16 software to document it. But i didn't find a meaningful usecase for this feature so i didn't add it to the esp software.

If you need it you can easily activate it by adding the following lines after line 241 to esp8266.ino
```
if (message == "TOOGLE_LIGHT")
{
  door.trigger_action(hoermann_action_toggle_light);
}
```

And publish a message with payload `TOOGLE_LIGHT` to your `SET_TOPIC`.

snoekieboe, 2021-05-22:
Thnx Stephan! I will add the code  and see if it works. Keep you posted!

stephan192, 2021-06-18:
I've added the feature but do not recommed to use it due to a missing feedback.

closed 2021-06-18 by stephan192

i5Js, 2021-07-25:
It's working; sending the payload(on/off) "LIGHT" turn on and off the light, but in HA, I don't know how to do correctly a switch like "Venting". I think It's because the value is always LIGHT.

stephan192, 2021-07-26:
You must configure your switch to work in optimistic mode (don't define a state_topic) and that's exactly the thing why i don't recommend using toogle light.

stephan192, 2021-07-26:
The complete solution would be:
1. Verify which bit of the broadcast status message is the light relais
2. Add an interface function to hoermann.cpp
3. Convert the cover position topic to a json message to be able to add more than one value
4. Adjust the configuration.yaml or additionally add [MQTT Discovery](https://www.home-assistant.io/docs/mqtt/discovery/)

**But i will not do this, at least in short term. Feel free to make a pull request.**