stephan192/hoermann_door #5: Promatic 4 - Bus scan fails

snoekieboe, 2021-05-30:
Hi Stephan, 

Sorry for bothering you but i've managed to source the PCB you designed and flashed both the ESP8266 and the PIC16 with your code to get it up and running in combination with my Hormann Promatic 4 door.

When i connect the PCD to the Hormann BUS:

* Initially nothing happens
* After toggling dip switch 8 (bus scan):
* The PCB is powered
* ESP8266 connects to the network (can ping the device)
* Lights on door start flahsing red/green rapidly (is supposed to happen on bus scan)
* As soon as bus scan finishes PCB loses power

Seems to my like the PCB is not recognized by the bus scan hence powering down the bus.

What could go wrong..?

* is the code not detecting the door?
* Is my PIC16 not flash succesfully?
* Something else?

any suggestions to point me in the right direction would be much appreciated

Thnx!

thomasdupas, 2021-05-30:
a promatic 4 has a different protocol (it's using plain modbus from what I understood), so that won't work with this variant for now.
I'm going to a friend in ~1 week to measure it with a scope/logic analyser, since it would be good to be able to do both variants of the protocol with the same device (with a different cable perhaps, depends on the pinout).
If successful I'll make a variant of the code which can do both, you can always contact me if you want follow-up on situation

snoekieboe, 2021-05-30:
Aaah that explains a lot!  I will keep in touch for sure! Thnx!

spacemanspiff2007, 2021-05-31:
@thomasdupas 
I am searching for a new garage door motor and haven't made up my mind yet.
It would be really nice if you could make your findings publicly available.

stephan192, 2021-05-31:
After a quick search i figured out that Hörmann offers an adapter called HAP1 HCP which makes it possible to connect old accessories (like the UAP1 which the PIC emulates) to new doors (like Promatic 4).
The old accessories are using version 1 of the HCP protocol whereas the new doors are using version 2 which isnt' compatible to version 1.
I found the following compatibility list: https://cdn.hoermann-cloud.de/fileadmin/_country/kataloge/pdf/hcp_kompatibilitaet.pdf

thomasdupas, 2021-06-04:
"progress", google revealed that someone already did some sniffing and has a working solution
https://github.com/hkiam/HCPBridge/blob/master/Investigation/ProtocolAnalyse.txt

What I find strange though, it wouldn't send distinct target states anymore, but more "button emulation"
for example:
0015: 02 17 10 70 00 03 01 02 10 00 00 00 00 00 00 00 00 00 00 6C 88 		&lt;- Tor auf pressed
0015: 02 17 10 78 00 03 01 01 10 00 00 00 00 00 00 00 00 00 00 6F 4A 		&lt;- Tor auf released

0015: 02 17 10 6C 00 03 01 02 20 00 00 00 00 00 00 00 00 00 00 9B 41        &lt;- Tor zu pressed
0015: 02 17 10 78 00 03 01 01 20 00 00 00 00 00 00 00 00 00 00 90 4A 		&lt;- Tor zu released

Sounds a bit odd to me why they wouldn't use the distinct states in the protocol for something like the UAP

closed 2021-07-26 by stephan192

pboschi, 2023-01-07:
> "progress", google revealed that someone already did some sniffing and has a working solution https://github.com/hkiam/HCPBridge/blob/master/Investigation/ProtocolAnalyse.txt
> 
> What I find strange though, it wouldn't send distinct target states anymore, but more "button emulation" for example: 0015: 02 17 10 70 00 03 01 02 10 00 00 00 00 00 00 00 00 00 00 6C 88 <- Tor auf pressed 0015: 02 17 10 78 00 03 01 01 10 00 00 00 00 00 00 00 00 00 00 6F 4A <- Tor auf released
> 
> 0015: 02 17 10 6C 00 03 01 02 20 00 00 00 00 00 00 00 00 00 00 9B 41 <- Tor zu pressed 0015: 02 17 10 78 00 03 01 01 20 00 00 00 00 00 00 00 00 00 00 90 4A <- Tor zu released
> 
> Sounds a bit odd to me why they wouldn't use the distinct states in the protocol for something like the UAP

Hi, did you ever finish the adaptation for ProMatic 4 series? Thanks