stephan192/hoermann_door #6: Error after upgrading HA

i5Js, 2021-07-24:
I've updated HA to the latest version, 2021.6.3, now, mqtt can't load due to the following message:

`2021-07-24 11:48:32 ERROR (MainThread) [homeassistant.components.homeassistant] The system cannot restart because the configuration is not valid: Invalid config for [cover.mqtt]: 'position_template' must be set together with 'position_topic'.. Got OrderedDict([('platform', 'mqtt'), ('name', 'garage'), ('command_topic', 'garage/cover/set'), ('state_topic', 'garage/cover/state'), ('qos', 0), ('retain', False), ('payload_open', 'OPEN'), ('payload_close', 'CLOSE'), ('payload_stop', 'STOP'), ('position_open', 100), ('position_closed', 0), ('position_template', '{% if value == "open" %}100{% elif value == "closed" %}0{% else %}10{% endif %}')]). (See ?, line ?). `

Any tips?

TIA

i5Js, 2021-07-24:
I've fixed adding the following:

`position_topic: "garage/cover/position"`

stephan192, 2021-07-25:
I did some changes on exactly this topic back in april.
My system is running without issues. Have you checked the lastest version of the [configuration.yaml](https://github.com/stephan192/hoermann_door/blob/main/homeassistant/configuration.yaml)?

i5Js, 2021-07-25:
Hi @stephan192, yes, I did, and I have had to add position_topic or it didn't work.

stephan192, 2021-07-25:
You're talking about the cover (lines 42-55 in my example), right?
In my example i have both `position_topic `and `position_template` set.

i5Js, 2021-07-25:
Yes, you're right, probably I took the example from the HA forum. I see in your example both are there and it's working fine.

Closing this issue.

closed 2021-07-25 by i5Js