stephan192/hoermann_door #7: Almost working with Supramatic E2

lg2dc, 2021-09-09:
Hi Stephan,

Firstly, thanks for your work and your sharing. I discovered that it was possible to draw boards ourself and even ask to build of a small amount of boards from manufacturers or web sites. (Even if I still have to learn a lot before being able to draw a working circuit myself :-D )

I got a board from a participant to [your HA Community Forum thread](https://community.home-assistant.io/t/hormann-garage-door-via-mqtt/279938).
My garage door is a SupraMatic E2, as far as I can understand from the Article Number on the motor box.

The participant was very kind to flash the PIC for me because I don't have any Microchip programmer.
And I flashed the ESP myself.

I can trigger commands: open, close, venting and toggle lights.
But _**the state is not reported**_, I always get "stopped" in the MQTT topic.

(The buttons on the motor box, the wall and remotes are still working as before, I do not have the error 6 like in the issue #3, but I don't have any UAP1 currently)

I'm quite newbie in hardware development and my first tests are with ESP32 dev boards using the [esphome.io project](https://esphome.io/).
So, to investigate the problem a bit deeper, I moved your code to an esphome project. I had to use a "custom component" as this implementation is not common off course. I succeed to reproduce the same behaviour: sending commands but no state reports.

I added some logs, especially inside the "parse_input" function, and discovered that the PIC is communicating with the ESP but it always reports 0 in the "state byte".

Here are the logs:
```
[12:32:52][D][garage:223]: parse_input rx_buffer[0] = 85
[12:32:52][D][garage:224]: parse_input rx_buffer[1] = 0
[12:32:52][D][garage:225]: parse_input rx_buffer[2] = 2
[12:32:52][D][garage:226]: parse_input rx_buffer[3] = 0
[12:32:52][D][garage:227]: parse_input rx_buffer[4] = 0
[12:32:52][D][garage:228]: parse_input rx_buffer[5] = 87
[12:32:52][D][garage:229]: parse_input rx_buffer[6] = 0
[12:32:52][D][garage:236]: parse_input 3 = 0
[12:32:57][D][garage:223]: parse_input rx_buffer[0] = 85
[12:32:57][D][garage:224]: parse_input rx_buffer[1] = 0
[12:32:57][D][garage:225]: parse_input rx_buffer[2] = 2
[12:32:57][D][garage:226]: parse_input rx_buffer[3] = 0
[12:32:57][D][garage:227]: parse_input rx_buffer[4] = 0
[12:32:57][D][garage:228]: parse_input rx_buffer[5] = 87
[12:32:57][D][garage:229]: parse_input rx_buffer[6] = 0
[12:32:57][D][garage:236]: parse_input 3 = 0
```

Do you have an idea of what could be problematic ?
Or where I should investigate more ?

I guess that the problem will be in the PIC code, but I wanted to confirm before going further.

I can share my esphome project code if it can help. I was planning to share it when it was ready.

Thanks in advance for your help!

stephan192, 2021-09-09:
Hi @lg2dc ,
i don't think there is a problem in the PIC code. You door is an E2 mine an E3. Maybe your door doesn't report back any status.
The door state (rx_buffer[3] & rx_buffer[4]) are 1:1 the original bytes received from the door drive. So i think you door always reports them as 0.

To dig deeper into the problem i recommend that you get yourself a RS485 USB dongle an log the messages on the hoermann bus the see what the door drive reports in the Broadcast status message.

lg2dc, 2021-09-13:
Hi @stephan192,
Thanks for your reply.

Right, there is a difference in the model. And that's why I was thinking that something should be adapted in the PIC code, like an "address" byte maybe.

If I well followed the code, the esp interface is continuously sending the content of the [broadcast_status](https://github.com/stephan192/hoermann_door/blob/0465f8ea44073dfaa37781686e3fe92cc14d4c3b/pic16/hoermann.c#L58) variable which is initialized to 0. Even if there is no change, right?

If I'm right, there is 2 possible cases:
1) the variable is never updated because the [broadcast message is not received or filtered](https://github.com/stephan192/hoermann_door/blob/0465f8ea44073dfaa37781686e3fe92cc14d4c3b/pic16/hoermann.c#L111)
2) the variable is updated with 0 contained in the broadcast message

So you are right, I should investigate what my drive is sending. Initially I was thinking to update the pic code to forward everything to the esp and log it, but there is maybe a better solution. Like listening directly the bus, as you suggested.

Do you think that the door drive will broadcast even if no UAP will be connected to the bus? Or should I connect in parallel the USB dongle and your board to the bus (I don't know if it's allowed for RS485)?

stephan192, 2021-09-13:
> If I well followed the code, the esp interface is continuously sending the content of the broadcast_status variable which is initialized to 0. Even if there is no change, right?

Yes, that's right.

> If I'm right, there is 2 possible cases:

My guess is **case 2**: Variable is updated with 0.

> Do you think that the door drive will broadcast even if no UAP will be connected to the bus? Or should I connect in parallel the USB dongle and your board to the bus (I don't know if it's allowed for RS485)?

You can (better to say must) connect the USB dongle in parallel. If you remove my board your door will not function anymore and report an error.
See https://community.home-assistant.io/t/hormann-garage-door-via-mqtt/279938/26

riogrande75, 2021-12-10:
@lg2dc Did you succeed creating a pic firmware working with your Supramatic E2?
I just noticed that the project was made for a E3 and I own two E2's. Got it working with raintonr's project, but it's unstable and I would prefer the hardware from this project.
I even bought a defective E2 motor drive, repaired it and use it now for testing - its much warmer in my lab than in the garage. 
Pls. let me know if I can contribute something for your investigations.

closed 2022-07-06 by stephan192

fox34, 2025-02-17:
Some background: The broadcast from a Supramatic E2 seems to differ from E3. It's just 1 byte of payload (uint8_t) instead of 2 (example: `00 01 01 C3`), which signals one of:
- Door fully closed (`0x02`)
- Door fully open (`0x01`)
- Door neither open nor closed (`0x00`)

There is no information about movement, light, error states etc. as in newer models. The code would need to be updated to also accept broadcasts with 1 byte (and to not provide and rely on incorrect/missing state information).

One other significant difference: The impulse command does not stop a moving door, emergency stop (signalling an error) must be used.