stephan192/hoermann_door #8: AOZ1038PI safe to work with 24V input?

mmanchev, 2022-01-28:
Hello, your project is great and it's exactlhy what I needed for my Hormann doors. I'm trying to reproduce the board, and while browsing for parts, I've noticed that datasheet for AOZ1038PI says "4.5 V to 18 V operating input voltage range". As per the schematic, the input voltage is 24V. Is it safe to operate in that range? Also the 10u C0805 capacitors should be at least 25V I suppose?

Looking forward for your reply.

Regards

stephan192, 2022-01-29:
Hi, it seems as if you have a twister in the part number. **The AOZ1038PI is not suitable**. The schematics contains the **AOZ1283PI** which is designed to operate up to 36V.
For C1 (the 10µF capacitor at the input) 25V is the absolute minimum rated voltage. If you have a closer look at datasheets for the different ceramic materials C1 should better have a rated voltage of 50V.

mmanchev, 2022-01-31:
Аhh, you are right. I used online gerber tool to generate me BOM with links to the componets, and probably because of stock shortage it linked to the wrong version. Thanks for that. I'm struggling to get all the components, but I will manage it soon or later. We can close this issue.

closed 2022-01-31 by mmanchev