stephan192/hoermann_door #10: Can I Substitute the  PIC16 for a different PIC16?

Gamogamer, 2022-08-15:
The chip you use in the design is basically sold out in all normal shops and I was wondering if I could Replace the 
PIC16F15324-I/SL
with the
PIC16F15325-I/SL

Looking at the Datasheet
https://ww1.microchip.com/downloads/en/DeviceDoc/PIC16F-LF-1532444-Data-Sheet-DS40001889C.pdf 
(from https://www.digikey.de/de/products/detail/microchip-technology/PIC16F15324-I-SL/6691276)
it seems the only difference is bigger memory. (Not that I really know what I am looking at otherwise)

I also had to redo the power circuitry for similar reasons and ill upload that in a fork or something once i figure out how to actually use github and that it works.

stephan192, 2022-08-16:
After a quick review of the datasheets i think (no guarantee) it should easily be possible to switch from the PIC16F15324-I/SL to the PIC16F15325-I/SL.
My conclusion is the same as yours. Both have the same peripherals, the only difference is that the 25's flash and RAM size is double compared to the 24. The most important thing to mention is that the PIC needs to have two UART modules.

But after a quick search i don't think it's easier to get the PIC16F15325-I/SL quicker than the PIC16F15324-I/SL.
In your place is would first try if i can get the same chip in a different housing or with a different temperature rating.
Mouser for example has currently some PICs in DIP package (PIC16F15324-I/P) on stock.
Have a look a page 556 (PRODUCT IDENTIFICATION SYSTEM) of the document you mentioned.

LocalNic, 2022-12-03:
> So as it turns out I overlooked that this is for Supramatic 3 and i have the 2. While my new design seems to work programmingwise and so on, I can't actually test it on a Supramatic 3. Bummer. If anyone wants the PCBs I am not looking to figure out a way to make them work with my V2s, nor do i have the equipment. I can also provide the schematics and so on if wished for.

Hi Gameogamer, I have two Supramatic 3 and I'm looking for a way to connect my DoorMotors with MQTT or something like that. but I am not so firm in programming the PIC. I have experience with the ESP anyway. 

If you still have the PCBs around leae me a line to 127.0.0.1@web.de

Gamogamer, 2022-12-12:
I wasn't able to send you an EMail, I get an error EMail not found

kom3n, 2022-12-13:
If still available, I am also insterested. Contact: borrob07@gmail.com

closed 2023-01-25 by stephan192

LocalNic, 2023-01-25:
Sorry I just received the notification by GitHub. Is misses a number 
127.0.0.11@web.de