stephan192/hoermann_door #12: Boards for sale

riogrande75, 2022-09-11:
I built 2 of these beauties with the boards from stephan192, before I noticed, that they don't work with my SupraMatic E2 doors.
Boards are fully assembled, FW is flashed, tested, so you just have to flash ESP32 with your settings again and you are good to go.
[image]

Jarda-cz, 2022-09-11:
Hello,
  I'm interested, if you have one left.

Abildgren, 2022-09-11:
Im also interested if there is one left.

riogrande75, 2022-09-12:
Ok, pls. send me your email adresses - I will contact you.

Jarda-cz, 2022-09-12:
My email is jarda.moravec@gmail.com . Thank you!

Abildgren, 2022-09-12:
> Ok, pls. send me your email adresses - I will contact you.

Github (a) Abildgren (.) dk

closed 2022-09-13 by riogrande75

johnreitan, 2022-11-02:
Any chance you could build a few more?

LocalNic, 2022-12-03:
I am also interrested .. I also like to oder them directly from jlbpcb. but for that I might need a short documentation which files I need to have the PCB right, the parts and their placement I guess. 

Or like my precessors, just buy it from some talented Maker. ;) 
br

MatthiasLohr, 2023-01-07:
> Any chance you could build a few more?

Same question from my side... Any chance?

wallo66, 2023-11-23:
I´m also interrested to buy a board, ist it possible?

threadstone83, 2024-02-07:
Me too. It would be nice if they could be bought.

realdimik, 2024-04-20:
I have 3 extra for sale. If someone is still interesting, send me your email for contact.
[image]

MatthiasLohr, 2024-04-20:
I'm interested, will write you an email shortly.

johnreitan, 2024-04-20:
I'm too, will email too shortly

MatthiasLohr, 2024-04-20:
Can't find your email address anywhere. In case you wanted our mail addresses instead: mine is mail@mlohr.com, would be great if you could contact me. Thanks!

realdimik, 2024-04-22:
Now all 3 are sold out.