stephan192/hoermann_door #13: Parts substitution & JLCPCB ordering

pboschi, 2022-11-26:
I was trying to produce a JLCPCB order list (see attachment), but the following parts are missing and very difficult to find. Are you able to suggest alternatives, or to modify the gerber with some more common ones?

The following should be ok:

- MCP1826S-3302E/DB ==> Ascend ASPL1117-3.3-DT-R
- ST485BDR ==> ST485EBDR
- ESP-12eSMD ==> ESP-12F(ESP8266MOD)
- 100u ==> RVT100UF16V67RV0120

But I'm having issues with the following:

- AOZ1283PI
- Can't be found.

- **SRN6045TA-220M** ==> **VLS6045EX-221M**
Lower max current, but is it needed? Also, i think this is a mistake; shouldn't the inductor be 22uH, reading on AOX1283PI datasheet?

- **PIC16F15324-I/SL** ==> ?
Candidates are:
PIC16F15325-I/SL but it can't be found
PIC16F18326-I/SL but it doesn't have 2 EUSART

I'm really not an electronic engineer and I don't have the skill to change the PCB, but I think that the buck regolator and PIC really need to be changed with more common ones.

[RS485Interface_top_cpl.csv](https://github.com/stephan192/hoermann_door/files/10096052/RS485Interface_top_cpl.csv)
[RS485Interface_top_bom.csv](https://github.com/stephan192/hoermann_door/files/10096053/RS485Interface_top_bom.csv)

pboschi, 2022-11-26:
I did some further checks, and those parts are available in other parts of the world... so I think I'll order from JLCPCB and just solder the missing parts by myself.
However, I still have a doubt about the 220/22uH inductor.

stephan192, 2022-11-26:
The inductor SRN6045TA-220M is a 22uH inductor.
See [Datasheet](https://www.bourns.com/docs/Product-Datasheets/SRN6045TA.pdf)

stephan192, 2022-11-26:
When i started this project back in 2019, all the parts were easily to source.
Sorry, but i will not update the gerber files (change parts or suggest alternatives) if i don't need it myself because the chip market is changing nearly every day and a part which is available today maybe sold out toworrow.

closed 2023-01-17 by pboschi