stephan192/hoermann_door #14: ProMatic 4 BiSecur 4510293

espuno, 2023-01-10:
does it also work with the Hörmann ProMatic 4 BiSecur 4510293 garage door opener?

Abildgren, 2023-01-10:
AFAIK Only with the HAP1 HCP adaptor https://www.tor7.de/hoermann-adapterplatine-hap-1-hcp as they have changed the protocol

stephan192, 2023-01-10:
The board only supports version 3 drives yours is a version 4 drive.
Check the lastest comments in [home assistant forum](https://community.home-assistant.io/t/hormann-garage-door-via-mqtt/279938/88) and [Thomas Dupas’s blog](https://blog.dupas.be/posts/hoermann-uap-hcp1/)

closed 2023-01-10 by stephan192

thomasdupas, 2023-01-11:
Hi, let me chime in here since it comes up often, since I'm indeed the author of above blog article (which has a blatant typo in the title/url, first time I notice it myself ;-)).

My intent/contribution at the time was to disclose the inner workings, as @Bouni had done at the time; where @stephan192 based his marvellous work here on, I myself build further on his work for own usage; and shared the research of E4 to the community. Thus coming full circle again, at least that was the intent.

I hope many people found the protocol information useful, that was my intent at the time. The focus seems to be on the finished product I have the impression, early on I got requests if I was willing to sell/ship readymade PCB's.
Since I had some spare PCB's for the minimal order (and to make the passive component picking/shipping price not greater than the passives) I indulged and shipped a few. I didn't anticipate the consequences this would have depending on the audience. It ends up at people with zero knowledge about homeassistent, mqtt, FTDI adapter, programming a microcontroller, … Some seem to expect help and aftercare/support on those items.
One bad experience in that area where I spent a considerable amount babysitting left a bad sentiment and made me abandon it.

Setting up something with that intent also has different requirements, both regarding to UX (buttons, visual feedback), as hardware (on-board FTDI chip, usb port, …), as documentation surrounding it; that becomes a commercial undertaking. On the homekit version some of those aspects are solved, less external dependencies, but if it’s not for own usage you have different license and compliance requirements from apple before you’re able to distribute it.
And on the other hand with the current supply chain it’s hard to make a static design and do small batches. You have to modify your BOM/design depending on the current stock, which will very likely be different if you do it again a month in the future.

So apologies for all the people who emailed me and didn’t get a reply, I literally made an email filter to park all those emails in a separate folder with the intent to go through them in due time, there are way over 100 requests in that folder I see now. I had a hardware redesign in progress at the time which I abandoned due to above reasons, the road to hell is paved with good intentions ;-).
But if it gets published it will come with a clear minimum expected self-sufficiency statement to avoid the same pitfall.

espuno, 2023-01-11:
People think your work is great @thomasdupas 
https://github.com/hkiam/HCPBridge/issues/13

Your PCB could you possibly make public?

TUNER88, 2023-01-11:
Hey @thomasdupas 👋 

It is really sad to hear that you have faced that bad experience. It is also fully understandable that you are unable to do any kind of support that was (for some reason) expected after shipping.

Would be cool if you could open source the PCB design files and the source code. The community seems to be quite big, I can imagine that we can find somebody who is able to update your BOM/design so we can order it ourselfs.
I do not have any selfmade PCB expirience yet, but would like to give it a try. There seems to be services that even do the assembly for you: https://www.pcbway.com/pcb-assembly.html

i5Js, 2023-01-24:
Hey @thomasdupas  I have just seen your post. Now I understand why I didn’t get any news from you. I sent you an email even today, because I flashed OTA the latest changes from this GitHub and now I can’t connect anymore with the board. I have tried to disconnect from my wifi to get the recovery mode, but It didn’t work either. I you could help me how to move on I would really appreciate. 

Anyway, thanks for your time you spent with me two years ago. 

Take care

Best