stephan192/hoermann_door #17: Unable to flash OTA

i5Js, 2023-01-25:
Hi all, 

I had my board flashed and working fine with my E4 door. I saw there were changes and I decided to flash the board OTA, like always. I made a mistake with the config.h and now the board doesn’t work, but, I can either find the way to reflash it OTA, all the ports are close now.  I guess I will need to solder the pins to have the serial working, but, what I did wrong? I remember, when I had this issues before, the board can enter in a recovery mode if it doesn’t detect the wifi, but not anymore… Any advice?

Thanks,

stephan192, 2023-01-25:
Hi @i5Js,
are you sure you got it working with an **E4**? If really you would be the first one ever. See #15
Which recovery mode, I don't know any recovery mode? If OTA isn't working you have to flash it via serial port.
Are you sure you opend your issue in the right Github project?

i5Js, 2023-01-25:
Ok, now all make sense. Thomas code and this code is different. His firmware had a recovery mode, and indeed, it worked in my E4. I will solder the serial port and will put his firmware back.

Thanks

closed 2023-01-25 by i5Js