stephan192/hoermann_door #18: Compatibility with Lineamatic gate motors?

fodi666, 2023-03-27:
Hi,
I've just stumbled upon this project. I have a Lineamatic 2 gate motor that I'd like to add to Home Assistant. Is this project compatible with this motor? I think it is a Series 3 motor, but I'm not sure. Can I use the "venting" option for the "small gate" option (partial open)? Does it also report open-close-opening-closing-partially open state?
Thanks

fodi666, 2023-03-27:
I've found a pdf somewhere about the compatibility of the HCP module where it said Series 3:
[image]
(unfortunately it's in Hungarian, but "3-as sorozat" means "series 3")
The module's manual also states the UAP1 or the UAP1-300 as a compatible addon board. Can it be used in this case then?

stephan192, 2023-03-27:
I can't say, because I don't have one. You have to try it yourself.

fodi666, 2023-04-03:
Thanks, I'll try. One more question: can the PIC and the ESP be programmed on board, or do they need to be programmed before soldering them? If they can be programmed on board I could order the board presoldered from China and it would make my job a lot easier. If not, then I'll have to leave at least the PIC and the ESP unsoldered.

stephan192, 2023-04-04:
They can both be programmed "on board".
To program the PIC connect a supported microchip programming tool to SV1.
To programm the esp connect a USB-UART converter to SV3 and enter "programming mode" like described in the README.md

closed 2023-08-29 by github-actions[bot]

kovlo, 2024-01-17:
> Thanks, I'll try. One more question: can the PIC and the ESP be programmed on board, or do they need to be programmed before soldering them? If they can be programmed on board I could order the [...]

Did you manage to try this on Lineamatic? I'd be interested in it as well. Were you able to order the board assembled and even preprogrammed? If yes, please share the source with me.

fodi666, 2024-02-26:
@kovlo no time so far, sorry

realdimik, 2024-04-21:
I have Lineamatic and yesterday I've connected a board to it.
It correctly reports status of the gate, even "option relay", but It can't send any command :(
I found that people had the same symptoms when they tried to use it with E4 ( https://community.home-assistant.io/t/hormann-garage-door-via-mqtt/279938/95 ). But I didn't change the pins, so it should be compatible with UAP 1, not with E4 series. I have currently no idea what to do next.

realdimik, 2024-05-03:
I've reviewed the code of Stephan and debugged it with Lineamatic. I've got that master (Lineamatic by itself) doesn't not send slave status requests.
After googling about the problem I found it in https://github.com/raintonr/hormann-hcp/issues/1 
In that issue Stephan by himself gave an advise to Robin regarding the problem reason

> Every message on the bus must start with a sync break (according to the LIN standard). A sync break are at least 13bits low, this is not the same as sending one or two zeros (each of them is only 8bits low). As the name says a sync break breaks the "normal" protocol frame to trigger an error = resync in every connected device.

I've reviewed code JS code of Robin and found that he followed the advice and solved the problem by switching boudrate=9600 and databits=7 temporary for sending the syncbreak.

Robin also mentioned that in the answer to Stephan:
> In fact, this can be achieved by setting the baud rate down to 9600, data bits to 7 then sending a zero, which in effect holds the transmit line high for long enough. Remembering to turn communications parameters back to normal before continuing.

I also found the same solution in one more repository https://github.com/steff393/hgdo/blob/master/src/uap.cpp#L209

But in the code of Stephan there is no such switching, just
>/* Start with a Syncbreak */
  tx_counter = 0;
  TX1STAbits.SENDB = 1;
  TX1REG = 0x00;

Unfortunately I have no experience with programming PIC controllers. By reading the code and documentation I've got how to change bound rate, but I have no idea how to change data bits. It would be great if someone experienced (or @stephan192 by himself) can help me with that.

stephan192, 2024-05-05:
Hi @realdimik,

my software for the PIC issues a LIN conform syncbreak. The implementation of Robin for examle is just a workaround which seems to be working but is not what the standard requests.

For the PIC you found the right piece of code where the syncbreak is generated. It are just three lines of code. Please have a look at the [datasheet (page 438 chapter 33.3.4)](https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ProductDocuments/DataSheets/PIC16F-LF-1532444-Data-Sheet-DS40001889C.pdf) for more details.

`/* Start with a Syncbreak */`
`tx_counter = 0;`
`TX1STAbits.SENDB = 1;`
`TX1REG = 0x00;`

**In conclusion: There is no need for baudrate change within my software!**

If the Lineamatic doesn't sent slave status requests i currently don't see any change to control the door via my board.

realdimik, 2024-05-05:
Hi Stephan,

thank you for the reply. It is clear.
My assumption was made based on that Robin got Lineamatic work, and I only needed to find the difference in approaches. But finally I found not a difference, but my misunderstanding of PIC registers that you've described. 

I will try to research a bit more. I will try to simulate Lineamatic by PIC and connect it to Hormann ese-868-bs that I have, and log how slave scan reply looks like.

In case of not success, I have plan B: connect a toggle gate button in Lineamatic board to an ESP.

For logging currently I'm collecting all messages in long memory buffer and make a breakpoint when it full to grab it using watches in MPLab. Does more standard way for logging exist?

rgds,
Dmitry

realdimik, 2024-05-05:
I've simulated Lineamatic by PIC and got from ese-868-bs a slave scan reply similar to UAP1, but with addr=0x29 and type=0x10 (makes sense). I've also measured timing. ese-868-bs replies by 9ms, not 3ms. 
Then I sent the same slave scan reply using PIC to Lineamatic, but didn't get any slave status request with any timing from 3ms to 12ms :(

stephan192, 2024-05-05:
What do you mean by
> I've simulated Lineamatic by PIC

The Lineamatic is the Master and the PIC or the ESE is the slave.

The "response delay" (3ms in my case) would be the next value i would start playing around, but you already did that. As far is a can remember is started with 0ms and my E3 didn't accept my slave can reply with 0ms delay.

My guess is that the Lineamatic doesn't accept/receive the slave scan reply by any reason. Normally the master keeps on sending the slave scan message until he receives a vaild reply. But, as far as i can remember, if it once has found a slave it never sends the slave scan again until the drive is reset.

I would do the following things as next steps:
1. Connect an additional rs485 dongle to the bus and log the messages on a PC.
2. Take an oszilloscope and measure the exact bus timings with the ESE.

realdimik, 2024-05-05:
> What do you mean by

I've rewrite the code in PIC to send the same packets with the same timing as master does (broadcast and bus scan only), connected PIC to ESE, and logged replies to catch a slave scan reply. And got it.

> As far is a can remember is started with 0ms

I haven't checked less than 3ms. I will do it.

> My guess is that the Lineamatic doesn't accept/receive the slave scan reply by any reason.

Mine is too.

> But, as far as i can remember, if it once has found a slave it never sends the slave scan again until the drive is reset.

It could help me. 30 minutes ago I got a slave status request. I changed logger to get more info, but I didn't get it again even after switching Lineamatic power off and on. 
_What do you mean by "drive is reset"?_

stephan192, 2024-05-06:
> What do you mean by "drive is reset"?
I mean performing a factory reset according to the drives manual. A power cycle is not enough.

14yannick, 2025-01-05:
Hi @realdimik,

Did you find the solution for your linematic? I get a code 4 after a couple seconds the board is connected looks like the rely is not getting received by the motor.

realdimik, 2025-01-06:
Hi 14yannick,

I've put it to pause. I had another stuff to do and I needed an oscilloscope and logic analyzer. Now I have them, but it is too cold outside to continue. I will return to it in the spring. 

In my case I have:
- Message BROADCAST_ADDR 0x00 well received
- I also can see CMD_SLAVE_SCAN 0x01 with sequence of different addresses (one by one)
- My device is replying to CMD_SLAVE_SCAN if address == 0x28 
- but I have no CMD_SLAVE_STATUS_REQUEST 0x20 from Lineamatic for my device

I debugged ese-868-bs, it is do the same but with other address and device type. And Lineamatic sends CMD_SLAVE_STATUS_REQUEST for it. I changed UAP1_TYPE and UAP1_ADDR to values from ese-868-bs in my device, but still had no CMD_SLAVE_STATUS_REQUEST

Factory reset of Lineamatic didn't help me.

I decide that problem is in the exact timing of reply in CMD_SLAVE_SCAN. It is the reason why I needed an oscilloscope.

What do you mean by "code 4"?

14yannick, 2025-01-06:
Hi thanks for your reply. I sold a board to someone having an LineaMatic H 2 he stated me following.

> It reads the status fine and I can issue commands. However, when the board is connected, the drive stops with error code 4 after a short, but seemingly random amount of time (maybe 1-5s). It’s basically what you’d expect if bit 4 of d1 isn’t set according to docs/hoermann.md. However, looking through the source code, that seems unlikely, and I wouldn’t expect the timing to be so random.

So he get's a bit further as his motor start to send the status slave request but then after some times it goes in protection mode (error 4) and stop working. He also suspect that the timing is not right. That the [3 ms delay](https://github.com/stephan192/hoermann_door/blob/08680a7c088993aadc717c87546d5804d2176dc4/pic16/hoermann.c#L221) are maybe causing the issue. Did you try to reduce it to 2 or 1 ms?

realdimik, 2025-01-06:
Hi, I've done many tests, including switching from 3ms to 0ms by 1, like that
[image]
No luck :(

stephan192, 2025-01-06:
Instead of decreasing i would suggest increasing the delay time for testing.
As far as i can remember, i started with 0ms but with 0ms my Supramatic E3 didn't accept the responses, because they came to quickly after the request.

14yannick, 2025-01-06:
based on the patent the reply from the slave should contain the address of the master.
In our code the address of the master is hardcoded as #define MASTER_ADDR 0x80.
has the LineaMatic maybe an other master_addr id?
In another repo I saw that the master can change depending the used equipment.

> // Master address: 128 (0x80) per default, 144 (0x90) for HAP1-HCP-Adapter

based on the patent the Master address should be at the beginning when the motor searching for connected devices. When you logged your buscan was the master address also 0x80?

realdimik, 2025-01-06:
> When you logged your buscan was the master address also 0x80?

It was 0x80. It was also one of mine assumptions.

> Instead of decreasing i would suggest increasing the delay time for testing.

It was a half year ago, but if remember correctly, I tried also.

14yannick, 2025-01-06:
Does anybody has access to an UAP?

realdimik, 2025-01-09:
I've recorded slave scan and reply with Supramatic E3 (because it is indoor :) ) with both devices: ese-868-bs and Stephan. Not yet analyzed, but the difference is visible. Maybe it will help.

ese-868-bs 
[image]

Stephan
[image]

realdimik, 2025-01-10:
The main difference is that for ese-868-bs tx and rx are always in apposite. For the Stephan's device it is not true in pauses between packages. 
@stephan192 do you know how to change that?

The next one - the length of syncbreak
- ese-868-bs has 0.934ms - exactly the same as Supramatic E3 by itself
-  the Stephan's device has shorter one 0.674ms

djasper-ha, 2025-01-10:
If it helps anything, I also ordered a board that is using https://github.com/Tysonpower/HCPBridgeMqtt_tynet which now comes with an E3 adapter. AFAIU, it uses an Esp chip with an RS485 sub-board for communication). The firmware for the E3 version uses an [unchanged fork](https://github.com/Tysonpower/hoermann_door/tree/main) of this repo (or rather @14yannick's) and works with no problem at the moment. So, I continue to suspect that there is some incompatibility in the PIC firmware, possibly related to the timing.

14yannick, 2025-01-11:
that are some goods new I'm also playing around with some out of the shelf. 
There are board from M5 Stack and Lilygo who have everything needed already included. 
[image]
Regarding you working board. The firmware has an autocorrect [function](https://github.com/14yannick/hoermann_door/blob/a15263402a5daeb43a0997aa02d5123089e884fa/esphome/components/uapbridge_esp/uapbridge_esp.cpp#L75). Could you check in your log if it's offen triggered.

realdimik, 2025-01-11:
Hello everybody.

I've solved the problem. 

I found that Hörmann doesn't use pullup resistors in ese-868-bs. I removed R4, R5 and R6 from Stephan's board.
Now it works well with both Supramatic E3 and Lineamatic!
It was the reason of the difference in pauses.

realdimik, 2025-01-11:
There is no equivalent for R5 in ese-868-bs. They have nothing at all.

stephan192, 2025-01-11:
That's very interesting.

R4 and R6 are so called bias resistors which should keep lines A and B in a valid range if no device is driving the bus. If the door drive itself already has them included and your lines are short enough removing them shouldn't be a problem. I don't remember why i choosed 20k a value. Maybe because it is way more then typically used on RS485 and so it shouldn't be a problem having them in parallel to possibly existing bias resistors inside the drive.

R5 is, as 14yannick already said, the termination resistor which should be placed, according to the specs, at both ends of the line. If the lines are short it may work with just one resistor. Has someone every measured the resistance between A and B of the door drive when everything is powerless and nothing is connected? Maybe the door drive has just 60Ohms build-in as termination between A und B. This could explain why you didn't find a termination resistor in the ese-868-bs.

djasper-ha, 2025-01-11:
> Regarding you working board. The firmware has an autocorrect [function](https://github.com/14yannick/hoermann_door/blob/a15263402a5daeb43a0997aa02d5123089e884fa/esphome/components/uapbridge_esp/uapbridge_esp.cpp#L75). Could you check in your log if it's offen triggered.

I don't see it triggered at all.

djasper-ha, 2025-01-11:
I also removed R4 and R6 and it didn't solve the problems I am seeing. The LineaMatic still stops after 1-3s showing error code 4.

14yannick, 2025-01-11:
Do you have some 4.7k Ohm resistor laying around you can put at R4 and R6. This would correspond to the resistance on tyson's board

djasper-ha, 2025-01-12:
Not currently, but I'll get some.

djasper-ha, 2025-01-17:
I tried with 4.7k Ohm as R4 and R6, but there wasn't any change in behavior. So, at least for this case it probably still is more of a software issue.

14yannick, 2025-01-18:
Chatgpt suggest following code change for the sync break.
```
/* Start with a Sync Break */
tx_counter = 0;
TX1STAbits.SENDB = 1;   // Enable Break sequence
TX1REG = 0x00;          // Write dummy value to initiate the break

/* Wait for SENDB to clear (Break complete) */
while (TX1STAbits.SENDB);  // Poll until SENDB is cleared by hardware

/* Write Sync Byte (0x55) */
TX1REG = 0x55;          // Sync byte follows the break

/* Enable Transmit Interrupt */
TX1IE = 1;              // Enable transmit interrupt

```
The while part looks interesting it may replace the arbitrary wait 2ms

djasper-ha, 2025-02-04:
I tried waiting for SENDB to clear, but it did not change the behavior. So, this probably needs more debugging.
(It's not high priority for me right now as I do have a working board)

14yannick, 2025-02-04:
I playing around making a Baudrate change in the PIC16 code as it's done in the esp. I will keep you updated as soon I got it working with my Supramatic E3 motor.

sebagsm, 2025-02-10:
hi! I got the tynet board. can someone pls help me how can I start the bus scan on the lineamatic2 ? thanks

sebagsm, 2025-02-10:
nope. so the adapter is needed. thanks for your help :) I will send them and email.

djasper-ha, 2025-02-10:
Yes. If you take a look at the compatibility table in the third comment above: https://github.com/stephan192/hoermann_door/issues/18#issuecomment-1485815572

Lineamatic and Lineamatic 2 both use the older variant of the bus protocol, shared with the Supramatic (E)3. For those, the adapter is necessary and you will also need a different firmware (I ordered mine with the E3 adapter and the E3 firmware pre-flashed).

sebagsm, 2025-02-10:
thanks for your help. this is the firmware in your board: esp32s3_tynet_BME280.bin ?

djasper-ha, 2025-02-10:
No, that's different. If you are interested:
esp32s3 - The s3 here is the flavor of the ESP32 chip (not the Supramatic 3 - yay abbreviations)
BM280 - This is if you want to attach (solder on) a BME280 temperature and humidity sensor.

What you need is the supramatic-e3.ota.bin (or supramatic-e3.factory.bin) from the top-level directory. Note that this is a EspHome based firmware and so you'll need to use that instead of MQTT to interact with the board.

sebagsm, 2025-02-10:
yes, thanks. I just flashed it.. I didnt know there is only esphome fw for supramatic. thanks

Tysonpower, 2025-06-18:
@kovlo i know that it works with lineamatic 2 motors from a customer :)

kovlo, 2025-06-18:
> i know that it works with lineamatic 2 motors from a customer :)

@Tysonpower Great, thank you! I am not sure what is my motor's version, but I will give it a try and I'll report what I find.