stephan192/hoermann_door #27: Supported devices

halacs, 2024-12-20:
Hi,

I have a ProMatic 3 door opener motor. Is it supported by this solution?

Thanks!

stephan192, 2024-12-20:
I'm not sure but according to https://github.com/stephan192/hoermann_door/issues/18#issuecomment-1485815572 i would guess NO.
Your door must be compatible with the "UAP 1" because my board emulates a "UAP 1".

14yannick, 2024-12-28:
Hi @stephan192 is right it's not compatible. I have a section in the [Document](https://docs.google.com/document/d/17D7XYupXnDYqXbOIze-K6jELF5pCdjmzDuHnc6SLtuA/edit?usp=drivesdk)  how you can have the functionality using a relay and reed sensors.

4920441, 2025-02-22:
Hi,

Hörmann hast at least two different systems it seems.... My Supramatic 4 needs the uap-1-HCP version:-(

https://www.tor7.de/hoermann-universaladapterplatine-uap-1-hcp

Is your project able to use that one too, somehow?

Why do they invent the wheel over and over again..... a garage door should be a very finite machine....

Cheers

14yannick, 2025-02-22:
Check [this](https://github.com/14yannick/esphome-hcpbridge) repo for uap-hcp. This need much easier hardware. You can get preassembled boards from tysonpower or I can also assist you.

4920441, 2025-02-23:
Hi,  thanks for that hint.

If someone iss interested in the board I've purchased, please let me now - its sits here as it was delivered.

I think there is no reason to modify it, when most of the components are not needed for my version, or is there?

Thaanks   a lot for clearyfying - really thought I hat a supramatic 3 in my garage......;-)

Cheers

closed 2025-06-01 by github-actions[bot]